fp64lib/fp64lib #3: Arduino 1.8.10 compile Error in Example "Double"

HaraldNoack, 2019-11-12:
Hello,

I wanted to test the Example "Double" on my Mega 2560 board, but I got some errors.

[ERROR_short.txt](https://github.com/fp64lib/fp64lib/files/3837949/ERROR_short.txt)
[ERROR_long.txt](https://github.com/fp64lib/fp64lib/files/3837950/ERROR_long.txt)

Thanks for your help!!

fp64lib, 2019-11-12:
Hi,

the code was not tested with a Mega 2560 board. However, the code should run, if you change line 33  of fp64def.h from
`#if !defined(__AVR_ATmega328P__) || defined(__AVR_TINY__) || !defined(__AVR_ENHANCED__)
`
to the following line
`#if (!defined(__AVR_ATmega328P__) || defined(__AVR_TINY__) || !defined(__AVR_ENHANCED__)) & !defined(ARDUINO_AVR_MEGA2560)
`
It seems like the __AVR_ENHANCED__ is not set for the Mega :-(

Hope that helps. Please leave me a note, if it does Then I will include that patch into the next release of fp64lib.

P.S: You'll find the fp64lib in the library folder in a standard installation of the Arduino IDE at c:\users\[username]\Documents\My Documents\Arduino\libraries\fp64lib.

HaraldNoack, 2019-11-13:
Hello,

problem solved and I get no "multiple copies of the fp64lib" Error.

[NO_ERROR_long.txt](https://github.com/fp64lib/fp64lib/files/3841660/NO_ERROR_long.txt)

THANKS !!!!!!

fp64lib, 2019-11-13:
Great, thanks for testing and replying. I'll include the patch into the next release. Have fun using fp64lib!

fp64lib, 2020-01-13:
The patch to support the Arduino Mega 2560 is now included in release 1.0.7., after the entire library has been successfully tested on a Mega2560 board.

closed 2020-01-13 by fp64lib