fp64lib/fp64lib #4: odd behavior of tan(pi/4)

gregh3285, 2020-01-11:
Running on an Arduino Duemilanove (compiled with the Arduino 1.8.5 IDE under Ubuntu linux 18.04, using version 1.0.6 of fp64lib), the following causes my Arduino to reboot:
// foo = tan(pi/4)
foo = fp64_tan(fp64_div(float64_NUMBER_PI,fp64_sd(4.0)));

The following seems to operate acceptably:
// foo = tan(pi/4 + 1e-6)
foo = fp64_tan(fp64_add(fp64_div(float64_NUMBER_PI,fp64_sd(4.0)),fp64_sd(1e-6)));

fp64lib, 2020-01-12:
Unfortunately, I was not able to reproduce the topic, at least with the current version of the Arduino IDE 1.8.10, neither under Windows 10 nor under Ubuntu 19.04, with version 1.0.6 of fp64lib. I suspect that the error is not directly coming from fp64_tan, but your code crashes some lines later. 

I propose to upgrade to the latest version of the Arduino IDE, as from 1.8.5 to 1.8.10 there were at least two updates on the avr toolchain. Then compile and upload the following minimal test program:
`
#include <fp64lib.h>
int led = 13; // pin for internal LED
char s[32];

char *toHex2( uint8_t n)  {
  static char res[3];
  uint8_t d = n >> 4;  res[0] = d + (d < 10 ? '0' : 'A'-10);
  d = n & 0x0f;   res[1] = d + (d < 10? '0' : 'A'-10);
  res[2] = 0;
  return res;
}

void dump( float64_t r ) {
  unsigned char *ur =  ((unsigned char *)&r)+7;
  Serial.print( "Result: ((float64_t)0x" );
  for( int i = 0; i < 8; i++ )
    Serial.print( toHex2( *ur-- ));
  Serial.println("LLU)\n");
}
void setup() {
  float64_t foo;

  Serial.begin(57600);
  Serial.println("starting test of fp64");
  pinMode(led, OUTPUT);     

  Serial.println("before tan"); Serial.flush();
  foo = fp64_tan(fp64_div(float64_NUMBER_PI,fp64_sd(4.0))); 
  Serial.println("after tan"); Serial.flush();
  dump(foo); Serial.flush();

  Serial.println("test finished");
}

void loop() {
  digitalWrite(led, HIGH);   delay(1000);
  digitalWrite(led, LOW);    delay(1000);
 }
`

This should produce the following output on the serial monitor:
`
starting test of fp64
before tan
after tan
Result: ((float64_t)0x3FEFFFFFFFFFFFC9LLU)

test finished
`

gregh3285, 2020-01-12:
Your code works just fine after I upgraded to 1.8.10.  I didn't even bother to try it under 1.8.5.  But, my code still doesn't work under 1.8.10.  If you'd be so kind, hold this issue open for a while to give me a chance to get to root cause.  I'm sure I'll have some embarrassing story to report when I'm done.  But, just in case...  Thanks for the quick response.

fp64lib, 2020-01-12:
Good to now  that at least fp64_tan is not the root cause. Good luck for hunting down the error. You might try simulavr and ddd (graphical frontend to gdb) for debugging - much better than all that Serial.print() debugging stuff. However, this will only work for parts of your code that don't need sensor input.

I am leaving the issue open, looking forward for reading your story...

gregh3285, 2020-01-12:
In my application I'm using U8glib.  I slowly starting moving relevant things over to your test application and came up with two cases--one that works and one that crashes.  The difference between success and failure depends upon the max_chars parameter passed into fp64_to_string().  If I use 14 or less, I'm fine.  If I use 15 or 16 (and likely more, but I didn't test that), my application reboots.  Again, running under Arduino 1.8.10 and version U8glib 1.19.1. In this case, you should be able to run this test app without a display (as long as you leave pins 3, 4 & 5 unconnected).  The documentation doesn't give a limit to max_chars.  I thought I was safe with 16.  Again, thanks for the feedback.  Good tip for simulavr or ddd.  I'll have to try them out.  Anyway, here's what I got to:
`
#include <fp64lib.h>
#include "U8glib.h"

#define DISPLAY_SCK (3)
#define DISPLAY_MOSI (5)
#define DISPLAY_CS (4)

U8GLIB_ST7920_128X64_4X u8g(DISPLAY_SCK, DISPLAY_MOSI, DISPLAY_CS);

#define BUFSIZ (32)
int led = 13; // pin for internal LED
char s[BUFSIZ];
float64_t foo;

#define MAX_CHARS_FP64_TO_STRING (14)  // works
//#define MAX_CHARS_FP64_TO_STRING (15)  // fails

char *toHex2( uint8_t n) {
    static char res[3];
    uint8_t d = n >> 4; res[0] = d + (d < 10 ? '0' : 'A'-10);
    d = n & 0x0f; res[1] = d + (d < 10? '0' : 'A'-10);
    res[2] = 0;
    return res;
}

void dump( float64_t r ) {
    unsigned char *ur = ((unsigned char *)&r)+7;
    Serial.print( "Result: ((float64_t)0x" );
    for( int i = 0; i < 8; i++ )
    Serial.print( toHex2( *ur-- ));
    Serial.println("LLU)\n");
}

void setup() {
    
    u8g.setColorIndex(1);         // pixel on
    
    Serial.begin(57600);
    Serial.println("starting test of fp64");
    pinMode(led, OUTPUT);
    
    Serial.println("before tan"); Serial.flush();
    foo = fp64_tan(fp64_div(float64_NUMBER_PI,fp64_sd(4.0)));
    Serial.println("after tan"); Serial.flush();
    dump(foo); Serial.flush();
    
    Serial.println("test finished");
}

void draw(void) {
    Serial.println("before tan, in draw()"); Serial.flush();
    foo = fp64_tan(fp64_div(float64_NUMBER_PI,fp64_sd(4.0)));
    Serial.println("after tan, in draw()"); Serial.flush();
    char *t = fp64_to_string(foo, MAX_CHARS_FP64_TO_STRING, 4);
    Serial.println("after fp_to_string, in draw()"); Serial.flush();
    strncpy(s, t, BUFSIZ);
    Serial.println("after strncpy, in draw()"); Serial.flush();
    u8g.setPrintPos(0,11);
    
    Serial.println("after setPrintPos, in draw()"); Serial.flush();
    u8g.print(s);
    u8g.setFont(u8g_font_profont12);
    u8g.setPrintPos(0,22);
    u8g.print("Hello World!");
       
    Serial.println("after u8g.print, in draw()"); Serial.flush();
}

void loop() {
    u8g.firstPage();  
    do {
        draw();
    } while( u8g.nextPage() );

    digitalWrite(led, HIGH); delay(1000);
    digitalWrite(led, LOW); delay(1000);
}
`

fp64lib, 2020-01-12:
I can now reproduce the error, even without u8glib with the following shorter program
--
#include <fp64lib.h>
#define S_BUFSIZ (32)
int led = 13; // pin for internal LED
char s[S_BUFSIZ];
float64_t foo;

char *toHex2( uint8_t n) {
  static char res[3];
  uint8_t d = n >> 4; res[0] = d + (d < 10 ? '0' : 'A'-10);
  d = n & 0x0f; res[1] = d + (d < 10? '0' : 'A'-10);
  res[2] = 0;
  return res;
}

void dump( float64_t r ) {
  unsigned char *ur = ((unsigned char *)&r)+7;
  Serial.print( "Result: ((float64_t)0x" );
  for( int i = 0; i < 8; i++ )
    Serial.print( toHex2( *ur-- ));
  Serial.println("LLU)\n");
}

void setup() {
  Serial.begin(57600);
  Serial.println("starting test of fp64");
  pinMode(led, OUTPUT);
  draw();
}

void draw(void) {
  Serial.println("before tan, in draw()"); Serial.flush();
  foo = fp64_tan(fp64_div(float64_NUMBER_PI,fp64_sd(4.0)));
  Serial.println("after tan, in draw()"); Serial.flush();
  dump(foo); Serial.flush();
  for( int i = 1; i < 18; i++ ) {
    Serial.print(i); Serial.println(":");
    char *t = fp64_to_decimalExp( foo, i, false, NULL );
    Serial.println("after fp64_to_decimalExp, in draw()"); Serial.flush();
    strncpy(s, t, S_BUFSIZ);
    Serial.println(s);
    t = fp64_to_string(foo, i, 4);
    Serial.println("after fp_to_string, in draw()"); Serial.flush();
    strncpy(s, t, S_BUFSIZ);
    Serial.println(s);
  }
  Serial.println("after strncpy, in draw()"); Serial.flush();
}

void loop() {
  digitalWrite(led, HIGH); delay(1000);
  digitalWrite(led, LOW); delay(1000);
}
--
It crashes exactly when the parameter maxDigits of fp64_to_string reaches 15 for this particular x of 0.999 999 999 999 993 89377... When called with 1 to 14 digits, the string representation will be rounded to 1 (or 1.00...E0 with fp64_to_decimalExp). But with 15 digits, the result should be 0.999999999999994. So, there is definitely some bug in fp64_to_string.

Thank you for detecting it. Sorry for the time you have to spent into hunting it down. I'll try to fix it as soon as possible.

BTW: Don't #define BUFSIZ, as this #define is already used by the stdio routines.

gregh3285, 2020-01-12:
It's open source.  When you find an issue, if you have time time, you look into it.  The library is really good stuff, so I'm happy to help.  Thank you for the work on the project.

fp64lib, 2020-01-13:
Bug is fixed in release 1.0.7. - thank you for reporting that nasty bug.

closed 2020-01-13 by fp64lib