fp64lib/fp64lib #5: Version 1.1.2 linker failures that 1.0.7 did not have.

gregh3285, 2020-02-08:
Version 1.0.7 of fp64lib links fine with my application.  When I upgrade version to version 1.1.2 of fp64lib (under Arduino IDE 1.8.10) I am receiving linker errors.  In my specific case, I see:

`/tmp/arduino_build_865983/libraries/fp64lib/fp64_addsf3x.S.o: In function '__fp64_add_pse_r0':
/home/gregh/Arduino/libraries/fp64lib/fp64_addsf3x.S:280:(.text.fp64lib.__fp64addsf3x+0xd2): relocation truncated to fit: R_AVR_7_PCREL against 'no symbol'
collect2: error: ld returned 1 exit status
Using library U8glib at version 1.19.1 in folder: /home/gregh/Arduino/libraries/U8glib 
Using library fp64lib at version 1.1.2 in folder: /home/gregh/Arduino/libraries/fp64lib 
exit status 1
Error compiling for board Arduino Uno.`

Attached is the code that fails.
[rpn_display_calc.ino.txt](https://github.com/fp64lib/fp64lib/files/4173752/rpn_display_calc.ino.txt)

fp64lib, 2020-02-08:
I am able to reproduce the error, also with a different file. Error does not occur when I link the library files directly with the source file, only when using the library. I have to deep dive into it, there sees to go something wrong with the code .sections

fp64lib, 2020-02-08:
I found it: when there where too many fp64lib, a local label in fp64add() was beyond the relative addressing range. Fixed by relocating the section more closely to the caller in V1.1.3.

closed 2020-02-08 by fp64lib

gregh3285, 2020-02-10:
Confirmed fixed.  Thank you for the good work.  Oder, danke für die Hilfe.  ;-)