fp64lib/fp64lib #6: input parameter stack clobber

gregh3285, 2020-02-10:
Looking at the assembler generated by the Arduino IDE, I get the sense that one of the fp64 functions is clobbering registers that the compiler expects won't get changed inside function calls.  When I define a constant 180 as a cast LLU, things fail.  If I go back and do the same thing by forcing fp64_lib to calculate that constant with fp64_sd(), things work fine.  Attached is a test case that demonstrates the failure mode.  Running the Ubuntu Linux version of 1.8.10 Arduino IDE and have upgraded to the 1.1.6 of fp64_lib.
[c180.ino.txt](https://github.com/fp64lib/fp64lib/files/4178345/c180.ino.txt)

fp64lib, 2020-02-13:
This was/is a tricky one.  It took me some time to figure out what is wrong, as I took quite some care that all the fp64lib functions do not unintentionally modify registers outside their parmeters. The problem ist, that they do modify the registres that are used to pass the parameters - which I thought is legit. 

However, the problem is - and it affects all functions with 2 or more parameters - that I do modify the registers that are used for passing the parameters, and gcc assumes that the parameters are not touched. In your case, it assumes that the constant C180 is still in r10 to r17. So, it works fine for the first iteration, but in all other iterations, C180 has a changed value.

I have to check the official ABI (application binary interface aka calling conventions). It C you are allowed to modify the parameters inside the called functions - maybe, I just have to supply some different keywords and/or #pragmas to bring gcc to no longer beleive that the second parameter is constant/untouched.

Thanks for bringing up that issue. It definitely limits the usability to nest expressions. Workarounds are to save intermediate results in variables and only nest functions with one parameter.

fp64lib, 2020-02-13:
Topic cannot be solved by keywords or #pragmas. avr-gcc documentation states clearly: "The requirement for the callee to save/preserve thecontents of these registers (i.e. r2-r17, r28-r29) *even applies in situations where the compiler assigns them for argument passing.*" A quick and dirty inserting of push/pop in fp64_div confirmed that, the problem does no longer occur. In order to do that properly, I have to rewrite some parts of the library, essentially all functions that use 2 or more parameters.

fp64lib, 2020-02-15:
Bug is fixed with V1.1.7 - I had to modify quite some routines and I tested nearly all routines for regsiter clobbering, both on a Arduino Pro Mini (with 16bit PC) and an Arduino Mega 2560 (with 24bit PC). Nesting of functions should now be possible without any negative side effects.

closed 2020-02-15 by fp64lib