fp64lib/fp64lib #7: Building with eclipse and avr-gcc linking error: relocation truncated to fit: R_AVR_13_PCREL against `no symbol'

dk5ee, 2020-03-03:
Using this library with Arduino IDE works out of the box. Nice work!

I tried compiling it with eclipse indigo, avr-eclipse plugin and avr-gcc 5.4 on debian buster arm64.

Eclipse project name is "float64test"

Simple main.c for a test:

```
#include <avr/io.h>
#include "fp64lib.h"

int main(void) {
	float64_t float1 = float64_NUMBER_PLUS_ZERO;
	while (1) {
		float1 = fp64_add(float64_NUMBER_ONE, float1);
	}
}
```

Every file from fp64lib was copied to the same directory as the main.c file.

Eclipse generated a makefile in ./Release :
```
################################################################################
# Automatically-generated file. Do not edit!
################################################################################

-include ../makefile.init

RM := rm -rf

# All of the sources participating in the build are defined here
-include sources.mk
-include subdir.mk
-include objects.mk

ifneq ($(MAKECMDGOALS),clean)
ifneq ($(strip $(ASM_DEPS)),)
-include $(ASM_DEPS)
endif
ifneq ($(strip $(S_DEPS)),)
-include $(S_DEPS)
endif
ifneq ($(strip $(S_UPPER_DEPS)),)
-include $(S_UPPER_DEPS)
endif
ifneq ($(strip $(C_DEPS)),)
-include $(C_DEPS)
endif
endif

-include ../makefile.defs

# Add inputs and outputs from these tool invocations to the build variables 
LSS += \
float64test.lss \

FLASH_IMAGE += \
float64test.hex \

EEPROM_IMAGE += \
float64test.eep \

SIZEDUMMY += \
sizedummy \

# All Target
all: float64test.elf secondary-outputs

# Tool invocations
float64test.elf: $(OBJS) $(USER_OBJS)
	@echo 'Building target: $@'
	@echo 'Invoking: AVR C Linker'
	avr-gcc -Wl,-Map,float64test.map -mrelax -mmcu=atmega328p -o "float64test.elf" $(OBJS) $(USER_OBJS) $(LIBS)
	@echo 'Finished building target: $@'
	@echo ' '

float64test.lss: float64test.elf
	@echo 'Invoking: AVR Create Extended Listing'
	-avr-objdump -h -S float64test.elf  >"float64test.lss"
	@echo 'Finished building: $@'
	@echo ' '

float64test.hex: float64test.elf
	@echo 'Create Flash image (ihex format)'
	-avr-objcopy -R .eeprom -R .fuse -R .lock -R .signature -O ihex float64test.elf  "float64test.hex"
	@echo 'Finished building: $@'
	@echo ' '

float64test.eep: float64test.elf
	@echo 'Create eeprom image (ihex format)'
	-avr-objcopy -j .eeprom --no-change-warnings --change-section-lma .eeprom=0 -O ihex float64test.elf  "float64test.eep"
	@echo 'Finished building: $@'
	@echo ' '

sizedummy: float64test.elf
	@echo 'Invoking: Print Size'
	-avr-size --format=avr --mcu=atmega328p float64test.elf
	@echo 'Finished building: $@'
	@echo ' '

# Other Targets
clean:
	-$(RM) $(FLASH_IMAGE)$(ELFS)$(OBJS)$(ASM_DEPS)$(EEPROM_IMAGE)$(S_DEPS)$(SIZEDUMMY)$(S_UPPER_DEPS)$(LSS)$(C_DEPS) float64test.elf
	-@echo ' '

secondary-outputs: $(LSS) $(FLASH_IMAGE) $(EEPROM_IMAGE) $(SIZEDUMMY)

.PHONY: all clean dependents
.SECONDARY:

-include ../makefile.targets
```

The output of 
/Release$ make clean && make all
is quite long, so I only paste the output of make all:

```
/Release$ make all
Building target: float64test.elf
Invoking: AVR C Linker
avr-gcc -Wl,-Map,float64test.map -mrelax -mmcu=atmega328p -o "float64test.elf"  ./fp64_10pown.o ./fp64_abs.o ./fp64_addsf3x.o ./fp64_asinx.o ./fp64_atan2.o ./fp64_atanx.o ./fp64_cbrt.o ./fp64_ceil.o ./fp64_classify.o ./fp64_cmp.o ./fp64_cmpA.o ./fp64_cmp_1.o ./fp64_cmpsd2.o ./fp64_copysign.o ./fp64_cosh.o ./fp64_debug.o ./fp64_disd.o ./fp64_divsf3x.o ./fp64_ds.o ./fp64_expx.o ./fp64_fdim.o ./fp64_fixxdfsi.o ./fp64_floor.o ./fp64_fma.o ./fp64_fmax.o ./fp64_fmod.o ./fp64_fmodx.o ./fp64_frexp.o ./fp64_fsplit3.o ./fp64_ftoa1.o ./fp64_gesd2.o ./fp64_getexp10.o ./fp64_hypot.o ./fp64_inf.o ./fp64_inverse.o ./fp64_isBzero.o ./fp64_isfinite.o ./fp64_isinf.o ./fp64_isnan.o ./fp64_ldb_1.o ./fp64_ldb_log2.o ./fp64_ldb_pi2.o ./fp64_ldexp.o ./fp64_log10.o ./fp64_logx.o ./fp64_lrint.o ./fp64_lround.o ./fp64_lshift64.o ./fp64_modf.o ./fp64_movAB.o ./fp64_movAC.o ./fp64_movBA.o ./fp64_movBC.o ./fp64_movCA.o ./fp64_mul64AB.o ./fp64_mulsd3x.o ./fp64_nan.o ./fp64_negdf2.o ./fp64_norm2.o ./fp64_pi2.o ./fp64_pow.o ./fp64_powserx.o ./fp64_powsoddx.o ./fp64_pretA.o ./fp64_pscA.o ./fp64_pscB.o ./fp64_round.o ./fp64_sd.o ./fp64_shift.o ./fp64_sidf.o ./fp64_signbit.o ./fp64_sinx.o ./fp64_sqrt.o ./fp64_square.o ./fp64_sssd.o ./fp64_stackA.o ./fp64_stackB.o ./fp64_stackC.o ./fp64_strtod.o ./fp64_swapAB.o ./fp64_tanh.o ./fp64_tanx.o ./fp64_tostring.o ./fp64_trunc.o ./fp64_zero.o ./fp64mulsf3.o ./main.o   
./fp64_modf.o:../fp64_modf.S:63:(.text+0xa): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
./fp64_modf.o:../fp64_modf.S:73:(.text+0x1c): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
./fp64_modf.o:../fp64_modf.S:91:(.text+0x3e): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
./fp64_modf.o:../fp64_modf.S:62:(.text.libfp64.asm.fp64_modf+0x6): relocation truncated to fit: R_AVR_7_PCREL against `no symbol'
./fp64_modf.o:../fp64_modf.S:62:(.text.libfp64.asm.fp64_modf+0x8): relocation truncated to fit: R_AVR_7_PCREL against `no symbol'
./fp64_modf.o:../fp64_modf.S:65:(.text.libfp64.asm.fp64_modf+0xc): relocation truncated to fit: R_AVR_7_PCREL against `no symbol'
./fp64_modf.o:../fp64_modf.S:68:(.text.libfp64.asm.fp64_modf+0x12): relocation truncated to fit: R_AVR_7_PCREL against `no symbol'
./fp64_modf.o:../fp64_modf.S:71:(.text.libfp64.asm.fp64_modf+0x18): relocation truncated to fit: R_AVR_7_PCREL against `no symbol'
./fp64_modf.o:../fp64_modf.S:73:(.text.libfp64.asm.fp64_modf+0x1c): relocation truncated to fit: R_AVR_7_PCREL against `no symbol'
./fp64_modf.o:../fp64_modf.S:86:(.text.libfp64.asm.fp64_modf+0x34): relocation truncated to fit: R_AVR_7_PCREL against `no symbol'
./fp64mulsf3.o:../fp64mulsf3.S:38: undefined reference to `__fp64mulsf3x'
./fp64mulsf3.o:../fp64mulsf3.S:39: undefined reference to `__fp64_round'
collect2: error: ld returned 1 exit status
make: *** [makefile:52: float64test.elf] Error 1
```

I am no expert in gcc internals, google gave some results, I tried some adviced like "-mrelax", but compilation was not successfull.

fp64lib, 2020-03-03:
I am currently struggling to get that makefile to run, as I miss some of the include files. Could you also upload the following generated files:
../makefile.init
../makefile.targets
sources.mk
subdir.mk
objects.mk

I already have an assumption. Could you run a compile run, where you insert the following statement in line 59 of fp64_fmod.S
`FUNCTION fp64_modf`
This might cure the segment overflow.

dk5ee, 2020-03-03:
Nice,
```
FUNCTION fp64_modf
```
solved the first error.

At first the missing files, then the new error.

There are no "makefile.init" and "makefile.targets" on my linux machine, this could be a hook to enter project specific make commands.

The .mk files are following:

objects.mk
```
################################################################################
# Automatically-generated file. Do not edit!
################################################################################

USER_OBJS :=

LIBS :=
```

sources.mk
```
################################################################################
# Automatically-generated file. Do not edit!
################################################################################

OBJ_SRCS := 
S_SRCS := 
ASM_SRCS := 
C_SRCS := 
S_UPPER_SRCS := 
O_SRCS := 
FLASH_IMAGE := 
ELFS := 
OBJS := 
ASM_DEPS := 
EEPROM_IMAGE := 
S_DEPS := 
SIZEDUMMY := 
S_UPPER_DEPS := 
LSS := 
C_DEPS := 

# Every subdirectory with source files must be described here
SUBDIRS := \
. \

```

subdir.mk
```
################################################################################
# Automatically-generated file. Do not edit!
################################################################################

# Add inputs and outputs from these tool invocations to the build variables 
C_SRCS += \
../main.c 

S_UPPER_SRCS += \
../fp64_10pown.S \
../fp64_abs.S \
../fp64_addsf3x.S \
../fp64_asinx.S \
../fp64_atan2.S \
../fp64_atanx.S \
../fp64_cbrt.S \
[... 266 lines cut ...]
	@echo ' '

%.o: ../%.c
	@echo 'Building file: $<'
	@echo 'Invoking: AVR Compiler'
	avr-gcc -Wall -g3 -gstabs -Os -std=gnu99 -funsigned-char -funsigned-bitfields -v -mmcu=atmega328p -DF_CPU=16000000UL -MMD -MP -MF"$(@:%.o=%.d)" -MT"$(@:%.o=%.d)" -c -o "$@" "$<" -mrelax
	@echo 'Finished building: $<'
	@echo ' '
```

These were autogenerated from eclipse. As you can see, in this case "-mrelax" is still in sources.mk, as I added that in the project configuration.

And the new error message from "make all" is:
```
./Release$ make all
Building target: float64test.elf
Invoking: AVR C Linker
avr-gcc -Wl,-Map,float64test.map -mrelax -mmcu=atmega328p -o "float64test.elf"  ./fp64_10pown.o ./fp64_abs.o ./fp64_addsf3x.o ./fp64_asinx.o ./fp64_atan2.o ./fp64_atanx.o ./fp64_cbrt.o ./fp64_ceil.o ./fp64_classify.o ./fp64_cmp.o ./fp64_cmpA.o ./fp64_cmp_1.o ./fp64_cmpsd2.o ./fp64_copysign.o ./fp64_cosh.o ./fp64_debug.o ./fp64_disd.o ./fp64_divsf3x.o ./fp64_ds.o ./fp64_expx.o ./fp64_fdim.o ./fp64_fixxdfsi.o ./fp64_floor.o ./fp64_fma.o ./fp64_fmax.o ./fp64_fmod.o ./fp64_fmodx.o ./fp64_frexp.o ./fp64_fsplit3.o ./fp64_ftoa1.o ./fp64_gesd2.o ./fp64_getexp10.o ./fp64_hypot.o ./fp64_inf.o ./fp64_inverse.o ./fp64_isBzero.o ./fp64_isfinite.o ./fp64_isinf.o ./fp64_isnan.o ./fp64_ldb_1.o ./fp64_ldb_log2.o ./fp64_ldb_pi2.o ./fp64_ldexp.o ./fp64_log10.o ./fp64_logx.o ./fp64_lrint.o ./fp64_lround.o ./fp64_lshift64.o ./fp64_modf.o ./fp64_movAB.o ./fp64_movAC.o ./fp64_movBA.o ./fp64_movBC.o ./fp64_movCA.o ./fp64_mul64AB.o ./fp64_mulsd3x.o ./fp64_nan.o ./fp64_negdf2.o ./fp64_norm2.o ./fp64_pi2.o ./fp64_pow.o ./fp64_powserx.o ./fp64_powsoddx.o ./fp64_pretA.o ./fp64_pscA.o ./fp64_pscB.o ./fp64_round.o ./fp64_sd.o ./fp64_shift.o ./fp64_sidf.o ./fp64_signbit.o ./fp64_sinx.o ./fp64_sqrt.o ./fp64_square.o ./fp64_sssd.o ./fp64_stackA.o ./fp64_stackB.o ./fp64_stackC.o ./fp64_strtod.o ./fp64_swapAB.o ./fp64_tanh.o ./fp64_tanx.o ./fp64_tostring.o ./fp64_trunc.o ./fp64_zero.o ./fp64mulsf3.o ./main.o   
./fp64mulsf3.o:../fp64mulsf3.S:38: undefined reference to `__fp64mulsf3x'
./fp64mulsf3.o:../fp64mulsf3.S:39: undefined reference to `__fp64_round'
collect2: error: ld returned 1 exit status
make: *** [makefile:52: float64test.elf] Error 1
```

My guess: the makefile from eclipse will compile all sources, and finds some quirks, whereas the arduino build process might only take those files in consideration, which are really used. Maybe this fixes some potential issues.

fp64lib, 2020-03-03:
The -mrelax is fine. Thanks for the updated files and for confirming that the relocation linker error is gone. I've found that in a couple of more files the FUNCTION statement is missing. This might also cause the new error, but I'll have to check and test that in the following evening(s).

fp64lib, 2020-03-04:
fp64mulsf3.S is a left over file of an old installation, the file was renamed to fp64_mulsd3x.S. If you remove the old file from your installation, everything will link fine. As I do not use Eclipse, I removed all references to it in the makefiles and compiling and linking worked without errors.

As I detected that segment allocation was also missing on some other files, it is best that you clean all files (except your project file and main.c) and install a fresh copy of the latest release, V1.1.9.

closed 2020-03-04 by fp64lib

dk5ee, 2020-03-04:
Thanks a lot, it's compiling now :)

The file:
 	fp64mulsf3.S
is still in the repository. After excluding this from the build, it works.
```
AVR Memory Usage
----------------
Device: atmega328p

Program:   11926 bytes (36.4% Full)
(.text + .data + .bootloader)

Data:        154 bytes (7.5% Full)
(.data + .bss + .noinit)
```

Now I can try this library out, yai! Thanks for the work :)