fp64lib/fp64lib #9: fp64_pow does not work as expected

brewmanz, 2021-05-21:
'normal' C/C++ function pow allows integer powers (both odd and even) for negative numbers. 
fp64lib function f_pow does not. 
The following test code snippet fails: 
```
  double x, y, expect, resStd, res64;
  x = -2.5; y = 2.0; expect = 6.25;
  res64 = f_ds(f_pow(f_sd(x), f_sd(y)));
  resStd = pow(x, y); res64 = f_ds(f_pow(f_sd(x), f_sd(y)));
  assertEqual(expect, resStd);
  assertEqual(expect, res64); // fails; f_pow returns NaN rather than 6.25
```

fp64lib, 2021-05-23:
Yes, that's right that the normal pow function allows integer powers. Due to the very limited capabilities of the Arduino regarding bitmasks, I did not include that, also to keep the code small. Including the check on integers would have more than doubled the code size for the fp64_pow function. 

That limitation was clearly documented at https://fp64lib.org/documentation/fp64lib-library-reference/.

closed 2021-05-23 by fp64lib

brewmanz, 2021-05-23:
If by 'clearly documented', you mean 'omitting information from a table', then yes. 
But not everyone has innate knowledge of all the edge conditions of mathematical functions regarding complex numbers. 
So I feel that, if you are omitting capabilities, if would be better to say something like 'BTW this does not handle negative integer powers, as handled by the standard pow function.'

fp64lib, 2021-05-24:
Knowledge of edge cases is not assumed at all, therefore they are all documented. But anyhow, I added a comment stating the difference - and I am thinking of an updated fp64_pow function...

fp64lib, 2021-05-29:
@brewmanz : Latest Version 1.1.14 now supports (-x)^y for integer powers. As expected, it is quite some bit-fiddling to determine if y is really an integer and whether it is odd or even. Code size has increased from 120 to 300 lines. Hope, it was worth the effort.

brewmanz, 2021-05-29:
Thank you. Perhaps introduce a 'pow_pos' function with shorter code just handling powers of positive numbers. IIRC unused functions do not contribute to final code size.