fp64lib/fp64lib #10: Possible issue with fp64_to_string

glennap01, 2021-08-31:
Hi,

Thank you very much for developing this library.

I'm quite willing to entertain the idea that I'm doing something wrong here but with the code:

`   float64_t A = fp64_atof("1.0");
   float64_t B = fp64_atof("2.0");
   float64_t C = fp64_atof("3.0");
   
   Serial.print("A:");
   Serial.println(fp64_to_string(A, 16, 15));   
   Serial.print("B:");
   Serial.println(fp64_to_string(B, 16, 15));  
   Serial.print("C:");
   Serial.println(fp64_to_string(C, 16, 15));  
   Serial.println("");  
    
   Serial.print("A:");
   Serial.println(fp64_to_string(A, 16, 15));   
   Serial.print("B:");
   Serial.println(fp64_to_string(B, 16, 15));  
   Serial.print("C:");
   Serial.println(fp64_to_string(C, 16, 15)); `

the output is:
`22:48:26.255 -> A:1
22:48:26.255 -> B:2
22:48:26.255 -> C:2
22:48:26.255 -> 
22:48:26.302 -> A:3
22:48:26.302 -> B:3
22:48:26.302 -> C:3`

might this be an issue with the fp64_to_string function or am I doing something wrong?

This is with an Arduino Mega, FP64lib version 1.1.4 and IDE 1.8.15

Thanks.

fp64lib, 2021-08-31:
Hi,
Thank you for that interesting case. I'll try to reproduce it in the next days. In the meantime, could you recheck, whether the same error occurs with the latest version 1.1.14?
I think, the error might be linked to the stack handling of nested function calls. If you make 2 statements out of your println, the error should also go away:
char *s = fp64_to_string(A, 16, 15);
Serial.println(s);
Regards,
Uwe

glennap01, 2021-08-31:
My Apologies - that was a typo.  I am indeed using the latest version - 1.1.14.  You are correct that if there is only one variable then it can be printed multiple times with no error or corruption.  The same issue occurs with two variables.

I am using nested function calls in other parts of the code with apparently no issue so I think it might be limited to the fp64_to_print function (at least in my experience).

glennap01, 2021-08-31:
Hi Uwe,
Sorry - I think I misinterpreted your response.  I tried this experiment:

` char *s;
   s = new char(40);

   float64_t A = fp64_atof("1.0");
   float64_t B = fp64_atof("2.0");
   float64_t C = fp64_atof("3.0");
   
   Serial.print("A:");
   s = fp64_to_string(A, 16, 15);
   Serial.println(s);   

   Serial.print("B:");
   s = fp64_to_string(B, 16, 15);
   Serial.println(s); 

   Serial.print("A:");
   s = fp64_to_string(C, 16, 15);
   Serial.println(s); 

   Serial.println("");

   Serial.print("A:");
   s = fp64_to_string(A, 16, 15);
   Serial.println(s);   

   Serial.print("B:");
   s = fp64_to_string(B, 16, 15);
   Serial.println(s); 

   Serial.print("A:");
   s = fp64_to_string(C, 16, 15);
   Serial.println(s); `

and it produced exactly the same incorrect result.  Sorry!

fp64lib, 2021-08-31:
Hi,
I was able to reproduce the error and narrow it down. There is something wrong when the 2nd parameter is 16. It works fine, if you use 17, i.e. fp64_to_string(A, 17, 15); With that, I do get the correct results 1, 2, 3, 1, 2, 3. Please use that as a work around until a bugfix is available.

BTW: If you want to have at maximum 15 (or n) digits after the decimal point (3rd parameter), then your second parameter should be at least 18 (or n+3) to have enough room for a leading sign,  a "0" and a decimal point.

Regards,
Uwe

glennap01, 2021-08-31:
HI Uwe,

Thanks for looking into it.  If I use:

`   float64_t A = fp64_atof("1.0");
   float64_t B = fp64_atof("2.0");
   float64_t C = fp64_atof("3.0");
   
   Serial.print("A:");
   Serial.println(fp64_to_string(A, 17, 15));   

   Serial.print("B:");
   Serial.println(fp64_to_string(B, 17, 15)); 

   Serial.print("C:");
   Serial.println(fp64_to_string(C, 17, 15)); 

   Serial.println("");

   Serial.print("A:");
   Serial.println(fp64_to_string(A, 17, 15));   

   Serial.print("B:");
   Serial.println(fp64_to_string(B, 17, 15)); 

   Serial.print("C:");
   Serial.println(fp64_to_string(C, 17, 15)); `

I still get 1, 2, 3, 3, 3, 3.  I also get the same result with 10,15.  I've worked around it for now by only converting to string once right at the end when the variables in question need to be displayed.  I'm happy to assist with any further testing.  

I think there is also an issue with fp64_modf which I'll post as a separate issue.  

I'm very grateful to you for developing this library.  I'm developing a telescope pointing application and without it the arduino would not have the precision required.

Regards,

Glenn.

fp64lib, 2021-09-01:
Hi Glenn,
error was found and corrected in V1.1.15. Hope you can now progress with your telescope pointing application. Owning a telescope, stellar and GPS calculations where also in my mind, when I started to create fp64lib. Great that you use it for that purpose. Good luck!
Uwe