fp64lib/fp64lib #11: Possible problem with fp64_modf

glennap01, 2021-08-31:
Hi again Uwe,

This may be a problem with fp64_modf, or perhaps I'm using it incorrectly.

With regards to the following code:

`   float64_t A = fp64_atof("1.23456789");
   float64_t B, C;

   B = fp64_modf(A, &C);
   
   Serial.print("A:");
   Serial.println(fp64_to_string(A, 15, 18));   

   Serial.print("B:");
   Serial.println(fp64_to_string(B, 15, 18)); 

   Serial.print("C:");
   Serial.println(fp64_to_string(C, 15, 18)); `

The result is:
09:02:20.949 -> A:1.23456789
09:02:20.949 -> B:0.0293209850788
09:02:20.949 -> C:3.27642282E-305

It appears to give strange results on every number I try.  Thanks again.

Regards,
Glenn

fp64lib, 2021-09-01:
Hi Glenn,
thank you foralways having a simple program for reproducing the error, helps a lot. I can confirm, that there is a bug in fp64_modf, that I did not detect up to now. As a workaround, you can use a combination of fp64_trunc and fp64_sub:

glennap01, 2021-09-02:
Hi Uwe,
Thanks for the update.  The least I can do is break it down to a simple demonstration given the amount of work you have put into the library.  If I sent you the 1200 lines of code it came from you would think I was crazy :).

Once I discovered the bug I created my own  two line modf function exactly as you suggested for a workaround.

Regards,
Glenn.

fp64lib, 2021-09-19:
Hi Glenn,
The bug with fp64_mod is fixed now, you can replace your two line modf function with fp64_modf. 
Regards,
Uwe

closed 2021-09-19 by fp64lib