fp64lib/fp64lib #12: atan2 broken

Huehnergott, 2022-01-19:
Hello,

I was using this lib on an arduino uno R3 and discovered, that for atan2 angles > 45° do not get calculated correctly. atan does what it should, as atan(1.00) equals 0.78... = 45°, whereas atan2 (0.5, 0.5) yields  strange angles around 343°. 

log: (azimuth is calculated as (atan2 (-y,-x)/Pi*180 +360 )mod 360)

02:36:01.823 -> Zeit 12:12:54
02:36:01.823 -> starTimeLocal 311.6393676127086
02:36:01.870 -> y -0.5016084x -0.5000116right asc / declination 101.53/-16.75
02:36:01.964 ->   azimuth / elevation 45.091/-47.3
02:36:02.853 -> Zeit 12:12:55
02:36:02.853 -> starTimeLocal 311.6435456646691
02:36:02.900 -> y -0.5016715x -0.4999826right asc / declination 101.53/-16.75
02:36:02.946 ->   azimuth / elevation 343.74/-47.29
02:36:03.836 -> Zeit 12:12:56
02:36:03.836 -> starTimeLocal 311.647723884726
02:36:03.883 -> y -0.5017346x -0.4999537right asc / declination 101.53/-16.75
02:36:03.929 ->   azimuth / elevation 343.75/-47.29
02:36:04.819 -> Zeit 12:12:57
02:36:04.865 -> starTimeLocal 311.65190176

fp64lib, 2022-01-20:
Hi Huehnergott,

found and fixed the bug, in some cases the registers were restored in the wrong order of operation. Should work now with V1.1.17.

closed 2022-01-20 by fp64lib

Huehnergott, 2022-01-21:
Dankeschön! Ich werd es ausprobieren.