fp64lib/fp64lib #13: fp64_to_string in some cases it does not work properly

giupa75, 2022-03-07:
The fp64_to_string function returns incorrect data if String () functions are used in the rest of the code in a different routine.
For the problem to occur it is not necessary that the routine with the "Killer" code be executed, but only that it be compiled and loaded into the uP.
Available to send you the code of my project for your study

fp64lib, 2022-03-15:
I tried several combinations with String() and fp64_to_string, and all worked. I have the assumption, that it has something to do with the static initialisation done in your code. fp64_to_string always return a pointer to the same memory area. Insert the following snippet to the beginning of your code:
`
String ffp64_to_string(float64_t x, uint8_t max_nr_chars, uint8_t max_leading_mantisse_zeros ) {
  const char* s = fp64_to_string(x, max_nr_chars, max_leading_mantisse_zeros );
  return String( s );
}
`

and replace your calls to fp64_to_string to calls to ffp64_to_string. This will create always a separate String for every call.

If that does not solve to topic, you ca send me a zip with your project to mail (at) fp64lib.org.

fp64lib, 2022-03-17:
Test program was more than 64kB in size and fp64lib was expecting it's constants to be in the lower 64kB of flash memory. With V1.1.18, code is now enabled to work with programs using more than 64 kB of flash memory.

closed 2022-03-17 by fp64lib

fp64lib, 2022-04-10:
V1.1.19 should finally fix the bugs when the program exceeds 64kB of flash memory.